UBCFormulaElectric/Consolidated-Firmware #971: Discussions Tab GitHub

Lucien950, 2023-09-14:
### Summary

I want a spot to float ideas without creating an issue.

### Deliverables

If you go into settings, the main page should have an option of setting up discussions.

gtaharaedmonds, 2023-09-18:
Hmm, what sort of ideas do you have in mind? Would making a thread in Slack serve this purpose, or setting up a meeting, and then creating an issue if the team is in agreement?

I don't like fragmenting information to lots of different sources if there isn't a good justification to do so. But if there are ideas that don't fit into existing discussion channels I'd consider enabling it.

Lucien950, 2023-09-18:
wait ur right slack is probably better

closed 2023-09-18 by Lucien950